FutureProofHomes/Satellite1-ESPHome #189: Get TAS2780 operational with 5V

that1guy, 2024-12-05:
**User Story:**
As a Sat1 user plugging my device into a typical laptop usb connection, I want my built-in speaker powered by the TAS2780 to work, so that I can flash the device and complete onboarding without having to switch over to a power-brick.

that1guy, 2025-02-06:
@that1guy & @gnumpi confirm this branch is functional, open PR for review.

https://github.com/FutureProofHomes/Satellite1-ESPHome/tree/TAS2780-5V

alextrical, 2025-02-17:
TAS2780 working on 5V on the Rev5 hardware, and can be run in the Public beta (Rev4.1) with a hardware modification, tying 5V to VBAT on the TAS2780

However the software is limiting the startup to only 5V from a USB-C to USB-C (USB PD) power supply, this should be changed to allow any 5V USB power source to enable the TAS2780

alextrical, 2025-02-21:
The modification to the hardware to allow running the TAS2780 is working well, however it has noticable noise/instability issues when 5V is fed through the VBUS to 5V regulator. We need to investigate another method for preventing backfeed and handeling of 5V or 9V+ input

that1guy, 2025-03-26:
@alextrical @gnumpi this ticket has been in review for a month.  What are we doing with this work please?  Thank you.

alextrical, 2025-03-26:
Its good to go from my point of view, Hardware is working at 5V to play back on the speaker

that1guy, 2025-03-27:
This ticket is moving back into ToDo.  We need the remaining work to be completed in order to close this ticket:

1. ESP Code must evaluate if the device is running at 9v or not and then set the TAS2780 registers accordingly.
2. Make sure the device runs stable when operating with 5v and powering on TAS2780 for audio playback.
3. Only release to production if #1 and #2 is complete and stress tested.

PR #297 (2025-05-07): Init TAS2780 with PWR_MODE0 if Voltage < 9V
- the TAS2780 is initialized even if the voltage is less than 9V (PowerMode set to 0),
   this give supported HAT boards the ability to drive a speaker at 5V
- for backwards compatibility the TAS2780 is still set to PowerMode 2 if the voltage is at least 9V 

closes #189
closed 2025-05-07 by that1guy