FutureProofHomes/Satellite1-ESPHome #198: BUG: Can't get BLE improv notification to pop up on iOS Device

that1guy, 2024-12-12:
Hey friends - I'm a bit nervous because none of my satellite1's are now detected for BLE improv on my iOS device.  Can anyone here test to confirm this is absolutely an issue on my end.

1. Remove your Sat1 completely from home assistant ESPHome
2. Factory reset the device
3. Flash the latest /develop fw on your Sat1
4. When device boots you'll eventually get warm white flickering LEDs indicating your device is in BLE improv and waiting for Wifi credentials.
5. Pull out your iOS device and open HA companion app.
6. Go to Settings -> Devices & Services -> Click "+ Add Integration" button
7. You should get a notification telling you a sat1 BLE device is ready for onboarding.

I can't get #7 to work for the life of me.  I must instead go to https://web.esphome.io/ and add the Wifi Credentials via serial improv to get the device on my Wifi and onboarded to Home Assistnat. 😱 😨

that1guy, 2024-12-12:
Pull down latest testflight companion app and test again to confirm.

that1guy, 2024-12-13:
This issue still persists.  Here's what I tried to narrow down the root cause:

1. Erased known good Sat1 hardware with `esptool.py -p /dev/cu.usbmodem2101 erase_flash | tee erase_flash.log`
2. Pulled down HA Companion app from Test flight.
3. Confirmed the companion app has Bluetooth permissions
4. Removed mmWave board from HAT

I even tested with Nabu hardware and other known good ESP32 devices.  I never get the popup that a new BLE device is detected for onboarding:

[image]

Here are my iPhone Specs:

[image]

that1guy, 2024-12-13:
Reached out to Nabu because we're stumped.  If this is issue with latests iOS version then everyone should be aware:

[image]

that1guy, 2024-12-13:
Using my iOS device to scan nearby bluetooth devices shows the Sat1 and Nabu hardware detected nearby.

[image]

gnumpi, 2024-12-19:
I get the same issue when building with ESPHome 2024.12.X ... 

@that1guy , can you confirm that you have this issue even with ESPHome 2024.11.3 ?

that1guy, 2024-12-19:
@formatBCE @ms1design @alextrical can you guys test the idea above for @gnumpi.  I'm still laser focused on completing video.  Ty!

gnumpi, 2024-12-19:
> [@formatBCE](https://github.com/formatBCE) [@ms1design](https://github.com/ms1design) [@alextrical](https://github.com/alextrical) can you guys test the idea above for [@gnumpi](https://github.com/gnumpi). I'm still laser focused on completing video. Ty!

Yeah that would be helpful , but if I recall correctly, you were the only one having that issue before, so the question is if you build your firmware locally and if you maybe already used e.g. the dev branch of ESPHome for building.

gnumpi, 2024-12-19:
Brad:

When I get to my local machine in the office I will confirm which version of ESPHome I am compiling with.  I’m unsure.

Are you saying the latest GHA beta build might solve the BLE improv problem?  Likewise are we sure it’s the ESP version we build with that even causes the issue?  Remember, this issue is occurring in ALL my ESP devices (Nabu hardware, EP1, and Sat1).

gnumpi, 2024-12-19:
for me this PR do cause the BLE-Improv issue: https://github.com/esphome/esphome/pull/7867

formatBCE, 2024-12-19:
> for me this PR do cause the BLE-Improv issue: https://github.com/esphome/esphome/pull/7867

It's not that pr. It's ESPHome 2024.12.0.

I have conversation with ESPHome devs on it.

ms1design, 2024-12-19:
I just flashed from `develop` branch (8db0e9ea0b895c37329bb2acc1c3412db2ebd4ee) and BT Improv works:

[image]

Logs:

[compile.log](https://github.com/user-attachments/files/18199329/compile.log)
[erase.log](https://github.com/user-attachments/files/18199328/erase.log)
[upload.log](https://github.com/user-attachments/files/18199327/upload.log)

gnumpi, 2024-12-19:
> > for me this PR do cause the BLE-Improv issue: [esphome/esphome#7867](https://github.com/esphome/esphome/pull/7867)
> 
> It's not that pr. It's ESPHome 2024.12.0.
> 
> I have conversation with ESPHome devs on it.

But #7867 is part of the ESPHome 2024.12.0 release, I just started with a the 2024.11.3 tag and only added these two commits to it:
```
commit 3c9aecd4f5aa3671fa76b018febef37d804b7172 (HEAD -> debugging_ble_improv)
Author: Jesse Hills <3060199+jesserockz@users.noreply.github.com>
Date:   Tue Dec 3 18:38:44 2024 +1300

    [ble] Allow setting shorter name for ble advertisements (#7867)
    
    Co-authored-by: Keith Burzinski <kbx81x@gmail.com>

commit 6ce93edb71b8d5a9953b7b0aa260dbb9b02d6645
Author: Jesse Hills <3060199+jesserockz@users.noreply.github.com>
Date:   Fri Dec 6 16:40:29 2024 +1300

    [esp32_ble] Use RAMAllocator to avoid panic abort from ``new``
```
Then improv via BLE stopped working.

formatBCE, 2024-12-19:
@gnumpi sorry, you're right, that PR caused it for 2024.12. :)
I see you're already on the thread in Discord. Let's hope for quick fix.

gnumpi, 2024-12-19:
quick summary of testing together with @ms1design and @alextrical:

- running our beta.0 firmware Ben had to update his HA to 2024.12.X to get improv working with his android phone.
- it worked for me and Miezko on iOS (both running HA already at 2024.12.X)
- we can also confirm that the develop branch, build with the ESPHome 2024.12.1 + PR #7984, works again as well

that1guy, 2024-12-19:
@gnumpi can we please build our latest fw with that version of ESPHome and graduate to beta release?  Let's make sure the manufactures have that .bin file.  🙏

gnumpi, 2024-12-19:
> [@gnumpi](https://github.com/gnumpi) can we please build our latest fw with that version of ESPHome and graduate to beta release? Let's make sure the manufactures have that .bin file. 🙏

I thought we want to stay with our beta firmware which was built with ESPHome 2024.11.2, the bug that will get fixed with PR #7984 got introduces only in 2024.12.0, so it is NOT in our beta firmware. Our next release will then be based on 2024.12.X including that fix.

that1guy, 2024-12-19:
@gnumpi I opted in my Sat1 into the beta track and upgraded it's firmware to [v0.0.5-beta.1](https://github.com/FutureProofHomes/Satellite1-ESPHome/releases/tag/v0.0.5-beta.1) via OTA.  

I then factory reset the Sat1 and removed the device from ESPHome and Home Assistant, however I still cannot onboard the device via BLE Improv.  This was the same issue that @ms1design had, right?

Is there a step I'm missing to get the [v0.0.5-beta.1](https://github.com/FutureProofHomes/Satellite1-ESPHome/releases/tag/v0.0.5-beta.1) firmware working?

gnumpi, 2024-12-19:
Are you running the latest HA version?

jazzmonger, 2024-12-19:
It works for me.  I'm using ESPHome version
[image]

I reflash from the DOCs site and ignore the Wifi setup.  I make sure I'm not connected to Wifi
[image]

Remove the ESPHome Sat1 Entity.
Go to Companion app and I get 
[image]

I'll try upgrading ESPHome and see what happens.

formatBCE, 2024-12-19:
@gnumpi does the onboarding work for you, or just improv itself?

gnumpi, 2024-12-19:
> [@gnumpi](https://github.com/gnumpi) does the onboarding work for you, or just improv itself?

just tested it again, onboarding works just fine here

that1guy, 2024-12-20:
Issue still persists with my recently flashed and factory reset Sat1 + my iPhone.. even with latest version of HA running.

Cache somewhere that needs to be cleared?

[image]

jazzmonger, 2024-12-20:
Have you deleted the HA companion app and reinstalled it?  A long shot, but who knows?

formatBCE, 2024-12-20:
Was about to suppose the same.
Also can you use other iPhone? And do you have Bluetooth on your HA? It would help in debugging.

that1guy, 2025-01-06:
Closing.  This issue has been resolved by Nabu here: https://github.com/esphome/esphome/pull/7984

closed 2025-01-06 by that1guy