FutureProofHomes/Satellite1-ESPHome #212: Document issues caused by weak wifi

that1guy, 2024-12-19:
**User Story:**
As a Sat1 User I want to know if my Wifi connection is stable and what issues I can experience if my wifi is bad, also how to debug the issue.

that1guy, 2024-12-19:
@ben-gineering db levels mean "you have weak wifi"?

In general, what are the issues a Sat1 customer can experience from bad wifi

@ben-gineering please share graphs that can help folks visualize the issue.

@gnumpi & @that1guy document how to enable debug mode.

ben-gineering, 2024-12-20:
testing log is here https://github.com/FutureProofHomes/Internal-Satellite1-Hardware/issues/140
from which we can then decide on what exactly should end up in the troubleshooting docs.

i found these during testing:
https://eyenetworks.no/en/required-good-wifi-signal-strength/
https://www.metageek.com/training/resources/wifi-signal-strength-basics/

i haven't double checked anything about their source, but at least both site agree and also reflect my experience when using esp32/esphome/home assistant:

-40dBm excellent
-50dBm good
-60dBm okay
-70dBm minimal viable signal strength, but might not be enough for music/voice streaming
-80dBm unreliable, might connect, but will cause problems

what kind of problems show up at what signal strength might be worth testing and documenting

closed 2025-04-23 by that1guy