FutureProofHomes/Satellite1-ESPHome #234: Sat 1 unable to connect to Hidden WiFi SSID

CyberGWJ, 2025-01-22:
## Describe the bug

Unable to connect to a hidden WiFi SSID

### Steps to reproduce

Setup a Sat 1 and try to connect to a hidden WiFi SSID using the HA app

### Expected behavior

Should be able to connect to hidden WiFi SIDD

### Environment
 - OS: Android 12

### Additional context

Would like to see support for 5Ghz WiFi networks

gnumpi, 2025-01-22:
Thanks, yes we haven't set the `fast_connect` parameter in our firmware:

https://esphome.io/components/wifi.html:
>fast_connect (Optional, boolean): If enabled, directly connects to WiFi network without doing a full scan first. This is required for hidden networks and can significantly improve connection times. Defaults to off. The downside is that this option connects to the first network the ESP sees, even if that network is very far away and better ones are available.

Let me do some testing and and discuss if we want to set this by default. 

In the meantime you could build your own firmware and add the fast_connect parameter manually to the wifi settings:
https://docs.futureproofhomes.net/satellite1-modifying-the-firmware/

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
```

We will add a full description on how to do that shortly.

CyberGWJ, 2025-01-22:
Thanks for the update.  I will look into building my own firmware.  Although since there may be a few changes coming since release I might wait.  The 5Ghz is minor.

closed 2025-02-21 by gnumpi