FutureProofHomes/Satellite1-ESPHome #236: Dynamically Detect & Initialize mmWave Sensor When Attached

that1guy, 2025-01-24:
**User Story:**
As a Sat1 user with an LD2410 or LD2450 sensor, I want the Sat1 to automatically detect when I install a sensor and display the entities in ESPHome, so that I don’t have to compile or flash a manual version of the firmware.

**Goal:**
- When an LD2410 or LD2450 is detected upon Sat1 boot, we should immediately initialize that sensor.
- We aim to avoid having separate versions of our firmware for different sensors, as this strategy does not scale with our highly “modular hardware goals.”

gnumpi, 2025-01-27:
The main problem here is to find a way on how to create the necessary components which get exposed to HA (i.e. sensors buttons etc) only when the device gets detected. As far as I know there is no direct support for this in ESPHome and I am not sure if this is even possible without changing its core functionality.

camcoding, 2025-01-30:
Is there any way in the configuration page (within HA) for the user to specify "Enable LD2410" or "I have LD2410 installed"?  I've tried to following the document:  https://docs.futureproofhomes.net/satellite1-presence-sensors/ but haven't been able to successfully get it to work.

that1guy, 2025-01-31:
> Is there any way in the configuration page (within HA) for the user to specify "Enable LD2410" or "I have LD2410 installed"? I've tried to following the document: https://docs.futureproofhomes.net/satellite1-presence-sensors/ but haven't been able to successfully get it to work.

@camcoding that would contradict goal #2 in this ticket and lead us down a path where we're managing multiple firmwares.  That is a slippery slope I'd really like us to avoid because there will be many permutations of hardware combinations a user may want for their Sat1 (based on our roadmap) and we really want to develop a better solution where all hardware combinations "just work".

that1guy, 2025-05-01:
This is a complicated task to achieve and may not be the ideal work to pursue at this time.

Alternative solution is to have Github Actions build variants LD2410 and LD2450 firmware at time of release.  Then to offer a more advanced ESP Web Tools experience that allows the user to check which firmware they want.

n8ohu, 2025-05-02:
I think that's the best option, even though it's not ideal. I know a couple of other projects do their own version of the ESPHome web installer with choices based on hardware used.

that1guy, 2025-05-13:
@gnumpi please work with @ms1design on the Webtools UI side of this project.  The UI should very simply ask the user:

Step 1: Ask the customer, "Which firmware do you want to install?"
- Production Firmware
- Beta Firmware

Step 2: Ask the customer, "Which optional mmWave sensor is attached to the Sat1?"
- No mmWave
- LD2410 mmWave
- LD2450 mmWave

We can use Lewis' work as a guiding light: https://docs.everythingsmart.io/s/products/doc/5c94342f-05dd-40ee-8808-072b80a8b7fc

The first version of this UI can be very simple and does not need to be scalable to support an infinite number of potential firmwares with an infinite numbers of questions.   It's okay if things are hardcoded.

Adding @ms1design to ticket.

gnumpi, 2025-05-16:
tracked in https://github.com/FutureProofHomes/Satellite1-ESPHome/pull/314

closed 2025-05-16 by gnumpi