FutureProofHomes/Satellite1-ESPHome #244: Guard implementation against no MWW (Micro Wake Word) in case user want to use OpenWakeWord instead or no wakeword.

Hedda, 2025-01-30:
## Summary

Guard implementation against no MWW (Micro Wake Word) in case user want to use [OpenWakeWord](https://www.home-assistant.io/voice_control/about_wake_word/) instead or no wakeword.

### Basic example

This is now implemented in home-assistant-voice-pe repository -> https://github.com/esphome/home-assistant-voice-pe/pull/291

### Motivation

Fact is some users prefer to use OpenWakeWord instead for now because creating your own MicroWakeWord model that works good is still hard

that1guy, 2025-01-31:
@Hedda thanks for making this recommendation.  It makes complete sense to do what we can to support on-device and server-side wake word.  I'll certainly look into this.

formatBCE, 2025-01-31:
The link you provided is related to the VA component, which SAT1 is using too.

The PE software, on the other hand, doesn't have any OWW-related setup. It just relies on MWW, unlike previous configs from NabuCasa (i.e. for ESP32-S3_BOX3, where you could choose which engine to use).

that1guy, 2025-02-21:
Just so we're clear: 

1. When [Sat1 firmware is updated to support the latest ESPHome release](https://github.com/FutureProofHomes/Satellite1-ESPHome/issues/256), we will get the feature [here](https://github.com/esphome/home-assistant-voice-pe/pull/291/commits/4bc81f6fef49549e4b252902c4e6adfa301e1f65) that enables users to disable MWW on the device.
2. Then we're saying the user should be able to use an HA voice pipeline that uses OWW on the server?

Thanks for confirming the above.

that1guy, 2025-04-23:
At this time we're not going to support server-side "OpenWakeWord".  Instead we will push forward with MicroWakeWord on the Sat1.

closed 2025-04-23 by that1guy

TheWebMachine, 2025-08-13:
Hello. Pardon me reviving an old request, but I just received my first SAT1 and was quite disappointed to find that I cannot use the oWW implementation I have been using on all of my other ESPHome mics.

May I ask why you have chosen not to add support for server-side oWW and if you would reconsider? Lack of oWW support precludes me from investing in additional SAT1 units as a replacement for my aging GHome speakers throughout my home, as I find mWW unusable with the very limited selection of models available. As previously mentioned, training custom mWW models is an often fruitless endeavor, hence why so few exist. Cramming these models into such limited hardware is a miracle unto itself. Those of us already running local LLMs like Ollama are likely also running Piper and OpenWakeWord to complete our offline services.

Many of us have invested heavily in custom oWW models because of their ease and versatility and support should definitely be included in the SAT1....eventually. Can you provide insight on the possibility of adding server-side oWW support in an upcoming firmware update?