FutureProofHomes/Satellite1-ESPHome #245: Factory Reset Doesn't Work After Beta Upgrade

camcoding, 2025-01-31:
## Describe the bug
After I upgraded my SAT1 to the latest beta version, if I factory reset the device the SAT1 displays that it is running version v0.0.4 but the beta toggle switch on the device is enabled (turned on).

### Steps to reproduce
On a new device shipped from FPH, once you follow the steps to connect the SAT 1 to Home Assistant then flip the beta toggle switch to on and click the restart SAT1 device.  After the SAT1 is up and running the beta software, factory reset it by holding down the right button for 25+ seconds and then follow the steps to add SAT1 to your HA system.  Open the SAT1 ESPHome tool and it will show your SAT1 is running v.0.0.4 but the beta toggle switch remains on (even if you attempt to switch it off).

### Expected behavior
After factor reset, I expected the SAT1 to revert to the factor version of v.0.0.4 with the beta flag turned off.

### Environment
 - OS: MAC OS

that1guy, 2025-02-15:
I will updated documentation to instruct how to solve this.  Thank you for pointing this out.  Solution coming shortly.

that1guy, 2025-02-16:
Instructs to do a full and complete reset of the sat1 have been added to documentation here: https://docs.futureproofhomes.net/satellite1-troubleshooting/#resetting-to-factory-settings

closed 2025-02-16 by that1guy