FutureProofHomes/Satellite1-ESPHome #246: LD2450 Setup did not install XMOS firmware (first time)

CyberGWJ, 2025-02-01:
## Describe the bug
The first time the adoption and install for the LD2450 did not install XMOS firmware.  Needed to install the debug feature for Wi-Fi and XMOS to do the XMOS firmware install.

Since it worked the second time it may not be a bug.  Although I did a complete erase and fresh install using ESPHome web dashboard.

### Steps to reproduce
See attached word document and log files.

### Expected behavior
A clear and concise description of what you expected to happen.

### Environment
 - OS: Windows

### Additional context
Attached is a Word document of the steps that were followed and a log file for the compile and installatio of the firmware.

[Sat1 setup second time after LD2450 install.docx](https://github.com/user-attachments/files/18628856/Sat1.setup.second.time.after.LD2450.install.docx)

[logs_satellite1-4e88e8_run.txt](https://github.com/user-attachments/files/18628859/logs_satellite1-4e88e8_run.txt)

gnumpi, 2025-02-01:
Hey, yes you are right what you experienced is the expected behavior at the moment. My motivation for disabling the automatic XMOS flashing when building a customized ESPHome firmware via the dashboard was that I assumed the user also wants to have full control over the XMOS updates in that case. We are aware that the documentation is not stating this clearly and we are working on a more user friendly way of handing the xmos updates from the ESPHome dashboard and will update the documentation accordingly!
Thanks for reporting your experience with that, I will keep you updated here!

CyberGWJ, 2025-02-01:
I am curious why the first time I did the adopt, compile and install it failed but the second time it worked.  Is it because I used ESPHome to load the initial firmware?

gnumpi, 2025-02-04:
That’s really a good question! So you are saying that the first time you manually initiated the xmos firmware via the debug button it did not work and when you reflashed the ESP and tried again, it did work?

CyberGWJ, 2025-02-04:
That is correct.  The only difference is that I use the ESPHome development website to do the install.  I documented the steps in the attached document above.

that1guy, 2025-02-15:
@CyberGWJ do you have any logs of what happened when it failed the first time?

that1guy, 2025-02-15:
Or can you reproduce the issue.  I've gone through these steps many times and have not stumbled upon this issue.

CyberGWJ, 2025-02-15:
@that1guy No I do not have logs from the first try.  I don't think I can reproduce it since I have loaded firmware for LD2450 on them.  If you want you can close it.  If it happens again I be sure to get logs.  Thanks

gnumpi, 2025-02-21:
The proper process for flashing the XMOS when building a customized ESPHome firmware via the ESPHome dashboard is still a work in progress and is being tracked in: https://github.com/FutureProofHomes/Satellite1-ESPHome/issues/243

that1guy, 2025-04-23:
Closing this ticket.  A feature will be implemented to automatically detect and support the mmWave in the future: https://github.com/FutureProofHomes/Satellite1-ESPHome/issues/236

closed 2025-04-23 by that1guy