FutureProofHomes/Satellite1-ESPHome #247: Cannot setup device on isolated wifi network

bentemple, 2025-02-03:
## Describe the bug
When powering the device after powering, device enters boot-loop if on a wifi network without internet access

### Steps to reproduce
1. Flash factory firmware to device
2. Connect device to an isolated wifi network without internet access
3. Pair device to ESPHome in Home assistant
4. Unplug and re-plugin device
5. Device enters infinite boot-loop

### Expected behavior
Device should function normally

### Additional context
I flashed the factory image via the ESPHome web installer, connected it to an isolated network with access to my HomeAssistant server
I added the device in the homeassistant ESPHome addon and verified all sensors showed up normally
Unplug power and power again
Device boots up and then is "lost" Lights stay white and you cannot factory reset the device with the button.
```
[15:54:56]ESP-ROM:esp32s3-20210327
[15:54:56]Build:Mar 27 2021
[15:54:56]rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
[15:54:56]Saved PC:0x42169a76
[15:54:56]SPIWP:0xee
[15:54:56]mode:DIO, clock div:1
[15:54:56]load:0x3fce3808,len:0x16c4
[15:54:56]load:0x403c9700,len:0xbc0
[15:54:56]load:0x403cc700,len:0x2e90
[15:54:56]entry 0x403c9950
[15:54:56]I (24) boot: ESP-IDF 4.4.8 2nd stage bootloader
[15:54:56]I (24) boot: compile time 15:29:51
[15:54:56]I (25) boot: Multicore bootloader
[15:54:56]I (26) boot: chip revision: v0.2
[15:54:56]I (30) boot.esp32s3: Boot SPI Speed : 80MHz
[15:54:56]I (35) boot.esp32s3: SPI Mode       : DIO
[15:54:56]I (40) boot.esp32s3: SPI Flash Size : 16MB
[15:54:56]I (45) boot: Enabling RNG early entropy source...
[15:54:56]I (50) boot: Partition Table:
[15:54:56]I (54) boot: ## Label            Usage          Type ST Offset   Length
[15:54:56]I (61) boot:  0 otadata          OTA data         01 00 00009000 00002000
[15:54:56]I (68) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[15:54:56]I (76) boot:  2 app0             OTA app          00 10 00010000 007c0000
[15:54:56]I (83) boot:  3 app1             OTA app          00 11 007d0000 007c0000
[15:54:56]I (91) boot:  4 nvs              WiFi data        01 02 00f90000 0006d000
[15:54:56]I (98) boot: End of partition table
[15:54:56]I (102) esp_image: segment 0: paddr=00010020 vaddr=3c170020 size=159790h (1415056) map
[15:54:56]I (365) esp_image: segment 1: paddr=001697b8 vaddr=3fc9d800 size=04974h ( 18804) load
[15:54:56]I (369) esp_image: segment 2: paddr=0016e134 vaddr=40374000 size=01ee4h (  7908) load
[15:54:56]I (372) esp_image: segment 3: paddr=00170020 vaddr=42000020 size=16bf58h (1490776) map
[15:54:57]I (647) esp_image: segment 4: paddr=002dbf80 vaddr=40375ee4 size=17910h ( 96528) load
[15:54:57]I (678) boot: Loaded app from partition at offset 0x10000
[15:54:57]I (678) boot: Disabling RNG early entropy source...
[15:54:57]I (679) cpu_start: Multicore app
[15:54:57]I (682) opi psram: vendor id : 0x0d (AP)
[15:54:57]I (687) opi psram: dev id    : 0x02 (generation 3)
[15:54:57]I (692) opi psram: density   : 0x03 (64 Mbit)
[15:54:57]I (697) opi psram: good-die  : 0x01 (Pass)
[15:54:57]I (702) opi psram: Latency   : 0x01 (Fixed)
[15:54:57]I (707) opi psram: VCC       : 0x01 (3V)
[15:54:57]I (711) opi psram: SRF       : 0x01 (Fast Refresh)
[15:54:57]I (717) opi psram: BurstType : 0x01 (Hybrid Wrap)
[15:54:57]I (722) opi psram: BurstLen  : 0x01 (32 Byte)
[15:54:57]I (727) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
[15:54:57]I (733) opi psram: DriveStrength: 0x00 (1/1)
[15:54:57]I (739) MSPI Timing: PSRAM timing tuning index: 5
[15:54:57]I (744) spiram: Found 64MBit SPI RAM device
[15:54:57]I (748) spiram: SPI RAM mode: sram 80m
[15:54:57]I (753) spiram: PSRAM initialized, cache is in normal (1-core) mode.
[15:54:57]I (760) cpu_start: Pro cpu up.
[15:54:57]I (764) cpu_start: Starting app cpu, entry point is 0x40377974
[... 112 lines cut ...]
[15:54:58][W][esp-idf:000]: I (2134) wifi_init: WiFi RX IRAM OP enabled
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: I (2136) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: I (2170) wifi:
[15:54:58][W][esp-idf:000][wifi]: mode : sta (d8:3b:da:40:79:e8)
[15:54:58][W][esp-idf:000][wifi]: 
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: I (2172) wifi:
[15:54:58][W][esp-idf:000][wifi]: enable tsf
[15:54:58][W][esp-idf:000][wifi]: 
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: I (2174) wifi:
[15:54:58][W][esp-idf:000][wifi]: Set ps type: 1
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: 
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: I (2177) wifi:
[15:54:58][W][esp-idf:000][wifi]: Set ps type 1
[15:54:58]
[15:54:58][W][esp-idf:000][wifi]: 
[15:54:58]
[15:54:58][W][component:157]: Component wifi set Warning flag: scanning for networks
Terminal disconnected: NetworkError: The device has been lost.
```

bentemple, 2025-02-04:
Quite possibly use error of course

gnumpi, 2025-02-04:
Hey, thanks for reporting! It is probably related to a known issue in ESPhome which causes the device to reboot when a connection timeout occurs. The device checks for firmware updates on startups. I will bump to the newest ESPhome version soon, I think it includes a fix for this issue.
But, your issue reminds me that we need to improve our offline firmware update strategy!

bentemple, 2025-02-05:
Was just looking into how I could standup the update server docker locally. Doesn't seem that hard but then I have to update the local url etc..

Sounds exactly like that but esphome bug. And super excited to hear you thinking about that as good support there will make me (and probably many other HA users) very happy.

closed 2025-04-23 by that1guy