FutureProofHomes/Satellite1-ESPHome #248: Wake word detection stops working.

KamilKeski, 2025-02-07:
## Describe the bug
After an indeterminate amount of time it stops responding to wake words.

### Steps to reproduce
No reliable repro steps atm.  After varying periods of inactivity wake words stop being recognized.  The device is still online and reporting to HA.  You can toggle led's, change different states, change wake words.  However it will not start responding to wake words until rebooted.

### Expected behavior
Continues to respond to wake words.

### Environment
 - OS: HAOS
 - None as this part is onboard.

### Additional context
Add any other context about the problem here.

Short log included below showing it going into idle, then never responding to a wake word.  But responding do other actions, including changing the wake word.

Satellite1 4e8898 Wake word changed to Hey Jarvis triggered by action Select: Select
8:08:12 AM - 10 minutes ago - Kamil
Satellite1 4e8898 Wake word changed to Okay Nabu triggered by action Select: Select
8:08:02 AM - 10 minutes ago - Kamil
Satellite1 4e8898 LED Ring turned off triggered by action Light: Turn off
8:07:57 AM - 10 minutes ago - Kamil
Satellite1 4e8898 LED Ring turned on triggered by action Light: Turn on
8:07:55 AM - 10 minutes ago - Kamil
Satellite1 4e8898 Wake sound turned on triggered by action Switch: Turn on
8:07:42 AM - 10 minutes ago - Kamil
Satellite1 4e8898 Wake sound turned off triggered by action Switch: Turn off
8:07:38 AM - 10 minutes ago - Kamil
February 6, 2025
Satellite1 4e8898 Assist satellite changed to Idle
1:17:27 PM - 19 hours ago

Zealand67, 2025-02-14:
Same behavior on my system. Usually after 24 hours after reboot stop responding to wake word. Reboot is only solution seems to help

that1guy, 2025-02-15:
This is a known issue we're trying to track down.

Please share any logs you can get your hands on that are relevant to this issue.

ddaudio1985, 2025-02-15:
I have a Sat1 sitting at my desk, I also had the same issue of the wake word stops responding. It would stop responding regardless of if it were a speaker or 3.5mm. When the speaker was connected It seemed to have gone longer without any issues but still did after a few days. With the external speaker (3.5mm) connected it seems that it would only last maybe a day or two then stop responding, but I noticed if I do not have a speaker  nor a 3.5mm connected it stays up and recognizes the wake word with no issue's. It has been up for about a week without it failing in that configuration. I hope this helps narrow down where the bug is.

Another observation is, when the Sat1 is used on a regular basis (multiple times a day) I didn't have the issue of the wake word not responding, only when I had to go into the office and left it sitting idle for a couple of days.

that1guy, 2025-02-18:
Quick update here: We are waiting on future variant of the Sat1 HAT from the manufacturer that enables us to more closely debug the XMOS with a JTAG connector.  Thank you for everyone's patience as we get the correct hardware on our desks to find the root cause.

tsx-5, 2025-03-02:
Thought I would add my own, not very scientific, observation.  I have two of these great devices - recently I placed one in the living room and one in my basement.  The living room one is driving a small speaker, the basement one is using the 3.5 mm audio out.  The most notable difference between the two locations is background noise - where the basement has a lot more computer fan noise.  What I have observed is that the one in the basement has shown this issue within a day, while the one in the living room has not shown it at all.

that1guy, 2025-03-04:
We are testing a firmware internally to try and isolate this issue.  Updates are coming.  Please drop any information here that can help with the troubleshooting.

ryanhindinger, 2025-03-14:
Didn't start to see this issue till I put the Sat1 in a case and "installed" it on top of our fridge in the kitchen.  Previous use was all testing so it got rebooted a lot, which may be why I hadn't seen it before.  I will add that in addition to not responding to a wake word, it will not play outbound TTS Audio (I have not tested if it will play ANY audio) and that is how I discovered it.  I was using Sat1 as a media player for a TTS announcement in my kitchen every morning and some mornings this week the automation would fire, the script would run, but nothing was announced.  And in the Sat1 device "Logbook" it clocked the device playing the audio then going back to Idle.  So, yeah--my "not an engineer" suspicion is the XMOS board taking an unreported nap.   Will add more here if I find it in that mode again.

gnumpi, 2025-04-18:
I finally traced the issue back to a problem during FreeRTOS task switching on the XU316.
Before I can dig deeper into the root cause or work on a fix, I need to address some general debugging reliability issues — they're currently limiting how much insight I can get when the problem occurs.

danwarren13, 2025-04-18:
Just wanted to add a comment that I have the same problem of my Sat 1 not responding to the wake word.  The only way to resolve it was to power cycle the unit.  I didn't know there was an updated firmware that  addressed the issue but Brad gave me some instructions on how to upgrade to a beta version XMOS v1.0.2.  I did the upgrade and it seemed to improve until a few days later when it quit responding again.  This time I pressed the mute button to mute/unmute and it would start responding.  Now, it still happens but much less frequently, maybe every other day.  So far I've been able to resolve it by pressing a button to wake it up except for about 2 times I had to power cycle it to recover.  I'm happy to provide debug info but I'm a noob to ESP so if you tell me what you need and how to compile it, I'd be happy to.  Thanks and good luck!

that1guy, 2025-05-06:
We're still hunting a ghost here everyone.  Please share any and all information possible if you find yourself dealing with this issue.  Thank you!

that1guy, 2025-06-10:
Our latest [v1.0.2 firmware release](https://github.com/FutureProofHomes/Satellite1-ESPHome/releases/tag/v0.1.2) has resolved this issue for 100% of brads unit.   Please test and respond back with your results.

gnumpi, 2025-06-12:
@rmeissn , are you running the latest ESPHome release? It would be indeed very interesting to understand what causes this issue on your end.

gnumpi, 2025-08-22:
Please reopen this ticket if anyone is still experiencing this issue also with the latest firmware release? Will close for now...