FutureProofHomes/Satellite1-ESPHome #256: The Latest ESPHome version 2025.2.0 Breaks Firmware

camcoding, 2025-02-19:
## Describe the bug
When attempting to recompile the firmware with the LD2410 line uncommented, ESPHomeBuilder within HA throws the following error message:  ValueError: Version pinning failed! Libraries esphome/esp-audio-libs@1.1.1 and esphome/esp-audio-libs@1.0.0 requested with conflicting versions!

### Steps to reproduce
1) Factory reset SAT1
2) Install Beta and reboot
3) Edit YAML and uncomment the LD2410 line, save, and attempt to compile/save to device wirelessly

### Expected behavior
Expected it to compile like it has done previously (prior to the ESPHome 2025.2.0 release)

### Environment
 - OS: HA is running HAOS on a Intel NUC

### Additional context
n/a

CyberGWJ, 2025-02-20:
FYI I get the same error with the base file.

Hedda, 2025-02-21:
Need to migrate firmware to use the new optimized ESPHome 2025.2.0 Audio Components and use latest esp-audio-libs libraries:

* https://github.com/esphome/home-assistant-voice-pe/pull/326

  * https://github.com/esphome/esp-audio-libs

    * https://github.com/esphome/esp-audio-libs/releases

Also see:

* https://esphome.io/changelog/2025.2.0.html

and

* https://github.com/esphome/home-assistant-voice-pe/pulls?q=is%3Apr+is%3Aclosed

Hedda, 2025-02-26:
In related news tetele is working on refactoring his ESPHome port for the Onju Voice satellite which understand faces similar hurdles:

* https://github.com/tetele/onju-voice-satellite/pull/108

Also depend on new i2s driver with full duplex support if hardware only uses a single i2s channel for both microphone and speaker?

* https://github.com/esphome/esphome/pull/8181

Possible workaround could be to use gnumpi's esphome_audio instead with "Dev next" PR patch submitted to his  repo by dimonb?

* https://github.com/gnumpi/esphome_audio/pull/79

formatBCE, 2025-02-26:
> Also depend on new i2s driver with full duplex support if hardware only uses a single i2s channel for both microphone and speaker?

Can you clarify? Respeaker Lite also has full-duplex single I2S, no problem with latest changes.

gnumpi, 2025-03-25:
Merged necessary changes to our firmware. It builds and passed first tests but still work in progress:
branch: prepare_for_esphome_2025_2

Hedda, 2025-03-26:
FYI, the upstream Home Assistent Voice PE fork has now been bumped to ESPHome v2025.3.0 now:

- https://github.com/esphome/home-assistant-voice-pe/pull/356

_It includes several changes that benefit the VPE:_

- _Fixes audio glitches when decoding certain FLAC files_
- _Reduces memory and CPU load when playing audio_
- _Increases the timeout for the server to send initial audio data (this was part of ESPHome v2025.2.2)_

- https://esphome.io/changelog/2025.3.0.html

that1guy, 2025-04-01:
The issue in this ticket is confirmed to be solved.  I have tested just now.  Simply

1. Run the latest code on your Sat1
2. Adopt your Sat1 to your ESP Builder Dashbaord
3. Modify the Sat1 firmware to uncomment the LD2410 firmware
4. Flash your custom firmware to your device and add your mmWave to the HAT.

The instructions are here too: https://docs.futureproofhomes.net/satellite1-presence-sensors/#firmware-installation

[image]

that1guy, 2025-04-01:
Opening back up.  This is the error we're getting with 2025.3.3 of ESPHome now:

[image]

n8ohu, 2025-04-01:
2025.3 seems to have broken this, as both of my tests produce the same results as above. One was with 2025.3.1 and the other was with the latest version. I don't currently have a backup with 2025.2.x, so I can't test on that.

PR #281 (2025-04-16): Adapt to breaking changes of ESPHome 2025.X
- [x] use Voice-Assistant PE components that have moved to official ESPHome repo
    - [x] FPH components have been adapted to the introduced braking changes
- [x] import remaining Voice-Assistant PE components from official repo instead of our fork 
- [x] adapt i2s components to support IDF5
- [x] adapt LED ring config for IDF5
- [x] set minimum ESPHome version to 2025.3.3
  - [x] updated requirements.txt
  - [x] updated GitHub actions

Note:
The official speaker-media-player component has a potential deadlock issue in the current version.
- [x] PR  [... cut]