FutureProofHomes/Satellite1-ESPHome #264: Regular XMOS firmware sensor updates

gnumpi, 2025-03-10:
## Summary
For testing purposes it is helpful to have the XMOS firmware sensor being updated regularly even after the XMOS has been detected successfully.

gnumpi, 2025-03-10:
Implemented in `loop_xmos_firmware_polling` branch.

that1guy, 2025-03-26:
@gnumpi I can see that @ben-gineering merged this into develop.  Can we close this ticket?

ben-gineering, 2025-03-26:
i merged develop onto the loop_xmos_firmware_polling branch because that was the combination i needed for long term testing. the xmos sensor in HA but also the newer xmos firmware.

if we want this xmos_firmware_polling in develop we could merge it, but i think @gnumpi can decide on that better than i can

that1guy, 2025-03-27:
Bring interval down to 10s and then merge with latest firmware.

PR #290 (2025-04-25): Update XMOS version sensor every 10s
Previously, the XMOS firmware version was read only once at boot time.
This change introduces periodic polling — the firmware version is now read every 10 seconds.

This serves as a lightweight heartbeat mechanism to monitor the XMOS's presence and responsiveness at runtime.

closes #264
closed 2025-04-25 by gnumpi