FutureProofHomes/Satellite1-ESPHome #275: Couple the Voice Satellite with existing speakers/tv's in the room for output and to 'mute' when making voice commands

TGrounds, 2025-03-22:
## Summary
I'm currently a user of the Echo and Echo Show devices and have Sonos as my whole-home audio platform.  A function that I really took for granted is that in my Alexa setup I identify the echo and Speakers that are in that room.  When I say the wake word, any audio playing from those sources automatically gets quieter so that I can have the conversation with the Echo device.  It would be great if the Satellite allowed me to do a similar pairing of speakers through Home Assistant rather than have to have the speaker attached directly to the Satellite.  It would also be optionally nice to direct the output from the Satellite to one of the speakers in the ceiling via software rather than a direct connection.

### Basic example
Include a basic example or links here.

### Motivation
I'm in the process of trying to remove the Echo devices from my home to replace completely with Home Assistant and your satellite voice assistant devices.  I can also see with the coupling of satellites within a room being able to just say play music and Jarvis knows the room that I'm in and can, as a default, play it where I am.  Just like it was able to turn on lights in the room without saying the room/device names.  I'm excited for the presence sensors as well so that I can get rid of my ESPresence devices (which help me find devices throughout the house) but also custom automation based on a person entering the room (lights, on, music on, TV on, etc.)  Same with turning things off when a person leaves.

that1guy, 2025-04-28:
Agree.  This is a great feature we can/should implement.

TGrounds, 2025-04-28:
Thanks for the consideration!  I love the work you all are doing and just received my first Satellite1 last week.  It's up and running now.  I certainly wouldn't put this ahead of the foundational tasks - but wanted to share a 'what if' dream in case it influences the implementation of those foundational tasks.  This is exciting what you are doing and I'm supporting your effort 100%

ItsHarper, 2025-04-28:
See the discussion at https://github.com/orgs/FutureProofHomes/discussions/30

TGrounds, 2025-06-27:
Thanks for the update!