FutureProofHomes/Satellite1-ESPHome #284: Media Player stops playing

gnumpi, 2025-04-18:
## Describe the bug
With changing to ESPHome 2025.4.0 (our current beta firmware) the media player is not working reliably anymore.
It stops with throwing: "The media pipeline's audio decoder encountered an error."

probably related to:
https://github.com/esphome/home-assistant-voice-pe/issues/310

### Steps to reproduce
Steps to reproduce the behavior.

### Expected behavior
A clear and concise description of what you expected to happen.

### Environment
 - OS: [e.g. Arch Linux]
 - Other details that you think may affect.

### Additional context
Add any other context about the problem here.

that1guy, 2025-05-06:
FYI as of 5/5/2025: 

[image]

rumlak, 2025-05-08:
[image]

For me it happens after only a few seconds of playing a track.  Every tenth attempt it may play for a minute.

gnumpi, 2025-05-13:
I think I tracked it down to an issue in the decoder pipeline (introduced with ESPHome 2025.2), after confirming I will create PR for ESPHome....

gnumpi, 2025-05-13:
https://github.com/esphome/esphome/pull/8776

gnumpi, 2025-05-14:
Let's keep track if PR #308 fixes the issue...

gnumpi, 2025-05-29:
Seems, that the fix did help. Closing this ticket...

closed 2025-05-29 by gnumpi