FutureProofHomes/Satellite1-ESPHome #285: Volume when Sat 1 reboots

CyberGWJ, 2025-04-19:
## Describe the bug
After updating to ESPHome 2025-04 and latest firmware, everytime the Sat reboots is resets the volume to 0 causing the red LEDs.  This is new to this firmware version.

### Steps to reproduce
Update to latest firmware and ESPHome 2025-04

### Expected behavior
Volume should remain at the last setting.

### Environment
N/A

### Additional context
None

that1guy, 2025-04-20:
@CyberGWJ I updated this ticket to clarify that this issue only appears on the latest BETA firmware.  Is that correct?  This issue is not happening in production, right?

Thanks.

danwarren13, 2025-04-21:
Just wanted to add my experience with the latest Beta firmware.  I updated and after reboot, the sound output was significantly reduced.  I didn't change a thing and I could barely hear it respond.  I had to turn the volume all the way up to hear it like I had it before (about 1/2 way?).  Seems like the sound output was cut by 1/2.

rumlak, 2025-04-26:
I am also having an issue with volume after installing the beta.   At 100% I can barely hear it if I am 12" away.

closed 2025-05-06 by that1guy