FutureProofHomes/Satellite1-ESPHome #288: Conversation Mode

wstewarttennes, 2025-04-24:
## Summary
I'd love a "conversation mode" where I say the wake word one time and then continues responding (like Chat GPT conversation mode) afterwards until i stop speaking for X seconds.  With the ability to plug into natural language models, the possibilities are endless for this device. But having to say Ok Nabu each time becomes annoying and repetitive in an actual conversation with it. If possible, we could also create new GPT or Llama, etc. "Conversations" for each of these, giving us a good record of past conversations.

### Basic example
User: **Ok Nabu,** what's the weather like today?
Nabu: The weather in Providence is 78 and sunny.
User: And what about tomorrow?
Nabu: No response
User: **Ok Nabu,** what about tomorrow's weather?
Nabu: Tomorrow in Providence will be partly cloudy with a high of 72.
User: Will I need an umbrella?
Nabu: No response
User: **Ok Nabu**, will I need an umbrella tomorrow?
Nabu: There's a 30% chance of light showers in the afternoon, so you might want to bring an umbrella just in case.
User: Can you add "buy groceries" to my to-do list?
Nabu: No response
User: **Ok Nabu,** add "buy groceries" to my to-do list.
...etc
### Motivation
This allows for a step from the Alexa days to a truly conversational home assistant.

ps. i'd be happy to build this but i'm not quite sure where to start (maybe it's possible directly in HA?). i'll do some more browsing...excited about this product

ItsHarper, 2025-04-25:
The latest beta has support for continuing conversations, but Home Assistant currently only activates that when the LLM response ends with a question mark. An option to always continue the conversation should probably be a feature request for them, I'd think.

that1guy, 2025-04-28:
Correct, our latest [v0.1.2-beta.0](https://github.com/FutureProofHomes/Satellite1-ESPHome/releases/tag/v0.1.2-beta.0) does support HA's continuous conversation feature (however, I recommend waiting until our next beta to resolve some unrelated media_player bugs we've recently resolved).

Internally, we are working on full-blown "realtime conversation" voice user experience (with interruptions).  Stay tuned.

wstewarttennes, 2025-04-28:
super cool thanks @that1guy @ItsHarper , i'll check out the Beta

rumlak, 2025-04-30:
Any idea when the next Beta is getting released?   I am trying to avoid a factory reset.

wstewarttennes, 2025-05-05:
@that1guy been playing around with the Beta. It's hard to know when it will or won't continue the conversation. Seems like only if the sentence ends with a question mark? But this is not an indicator of a need to continue conversation imo. My ideal situation would be a "Wake Word" and a "Sleep Word" that would put it in continuous conversation mode until you say the Sleep Word? I don't fully understand what we can/cannot control in this repo vs home assistant so not sure if this is the right place to build

that1guy, 2025-05-06:
> Any idea when the next Beta is getting released? I am trying to avoid a factory reset.

@rumlak the latest beta is out already. :)

> [@that1guy](https://github.com/that1guy) been playing around with the Beta. It's hard to know when it will or won't continue the conversation. Seems like only if the sentence ends with a question [...]

@wstewarttennes that functionality won't happen with the HA pipeline to my knowledge, but we're working on exactly this functionality for our upcoming [AI Base Station](https://futureproofhomes.net/pages/ai-base-station).

wstewarttennes, 2025-05-15:
> [@wstewarttennes](https://github.com/wstewarttennes) that functionality won't happen with the HA pipeline to my knowledge, but we're working on exactly this functionality for our upcoming [AI Base Station](https://futureproofhomes.net/pages/ai-base-station).

That is awesome, thanks for sharing!! Any Beta offerings over there? The conversation mode has me thinking of tons of ways to try to integrate voice into my day-to-day. I work in [food-tech](https://cityflavor.com) building ordering systems that tie into local merchants. I'm interested as a side project of integrating the ordering into my local pipeline so i can order my favorite meals for pickup/delivery. Would love to try to build something integrated with this repo or maybe it's as simple as an MPC server integration.