FutureProofHomes/Satellite1-ESPHome #291: LEDs don't work

wjschmitt, 2025-04-26:
## Describe the bug
Most of the LEDs in the ring stopped working.  Currently only two will light up.  When I examine the circuit board, it appears that many of the LEDs have fallen off.

### Steps to reproduce
Plug in power and turn on LED ring in HA.

### Expected behavior
I expect to see an entire ring of LEDs, which I saw when first powering on the board.

### Environment
 - OS: Windows
 - Other details that you think may affect - I have placed the board into the enclosure several times to check the fit.  Maybe doing this caused the tiny LEDs to be dislodged.  If so, the board is not very robust.

### Additional context
Add any other context about the problem here.

that1guy, 2025-04-26:
Absolutely looks like damaged product.  These things happen and we're working on getting manufacturing defects down with our new manufacturers we've just migrated to.

Please go to account.futureproofhomes.net to request an exchange and we'll have known good product sent back to you.

Thank you!

that1guy, 2025-04-26:
Closing this ticket since we're processing return.

closed 2025-04-26 by that1guy

wjschmitt, 2025-04-26:
Thanks for the quick response.  Exchange initiated.