FutureProofHomes/Satellite1-ESPHome #293: Use the TAS2780 Mini DSP

that1guy, 2025-04-28:
## Summary
The Satellite1 has a TAS2780 amplifier onboard whose potential is untapped.  Specifically, the amplifier comes with a mini-dsp built-in.  I recommend we give the customer presets they can select from in a drop-down in the ESPHome firmware.  Those choices should be filtered by the current USB-C Power Draw the Sat1 is pulling from the power supply (which we can use to assume the quality speaker they're running).

### Basic example
[TAS2780 Data Sheet](https://www.ti.com/lit/ds/symlink/tas2780.pdf?ts=1745810706809&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTAS2780)

### Motivation
The Sat1 amplifier should be able to maximize the users listening experience no matter if they're running a small, medium or large speaker in our official enclosures.

gnumpi, 2025-04-28:
Can you be more precise, what you want to achieve here? Which features are you referring to?  Which settings would these presets adjust? Can you explain your thoughts on filtering by the negotiated power supply?

that1guy, 2025-05-02:
Closing this ticket for the more focused and detailed ticket we have here: https://github.com/FutureProofHomes/Satellite1-ESPHome/issues/298

closed 2025-05-06 by that1guy