FutureProofHomes/Satellite1-ESPHome #304: Satellite reboots randomly

galoki, 2025-05-08:
On factory firmware or beta firmware, the unit is randomly rebooting.    I was also unable to download logs when connected to PC via USB.  While troubleshooting, I removed the presence sensorHLK-LD2410B-P brand is Jessinie.  Random reboots stopped and I was able to access full functionality of the usb connection to pc and download logs.

galoki, 2025-05-13:
updated to latest beta and "took control", after that no reboots

galoki, 2025-05-19:
so. that didn't fix it.  just extended the time between reboots.

galoki, 2025-05-21:
I swapped hats and cores between the working unit and rebooting unit.  maybe this will identify which component is causing the reboot.

galoki, 2025-05-23:
reboot problem followed the HAT.

that1guy, 2025-06-27:
@galoki If the issue persists @ v0.1.2 then please ping me on the [#community-support channel](https://discord.com/channels/1242282357298298941/1323128892814397500) and I'll get a replacement Sat1 your way.