FutureProofHomes/Satellite1-ESPHome #318: Can not load the config from esphome.

rikksullenberger, 2025-05-17:
## Describe the bug
When I try to take control of the device in esphome and reflash it I get an error. I have read the documents

### Steps to reproduce
took control, changed the package to the specified github location, hit install

### Expected behavior
It should have compiled, then flassed OTA

### Environment
Home assistant 25.5
Esphome 2025.4.2

### Additional context
I get this error : 

INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/satellite1-929968.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Error reading file /config/esphome/satellite1.base.yaml: [Errno 2] No such file or directory: '/config/esphome/satellite1.base.yaml'

ubstitutions:
  company_name: FutureProofHomes
  project_name: Satellite1

  #set this version by GHA at build time
  esp32_fw_version: dev

packages:
  FutureProofHomes.Satellite1: github://futureproofhomes/satellite1-esphome/config/satellite1.yaml@develop
  # This is an inline package to prefix the on_client_connected with the wait_until action
  # It must appear before the actual package so it becomes the orignal config and the
  # on_client_connected list from the package config is appended onto this one.
  va_connected_wait_for_ble_and_flasher:
    voice_assistant:
      on_client_connected:
        - wait_until:
            not: ble.enabled
        - wait_until:
            not: memory_flasher.in_progress
        - delay: 2s
    
    wifi:
      on_disconnect:
        - ble.enable:
      on_connect:
        - delay: 5s  # Gives time for improv results to be transmitted
        - ble.disable:
    
    satellite1:
      on_xmos_no_response:
        then:
          - if:
              condition:
                - lambda: return id(init_in_progress);
                - lambda: return id(xflash).has_image_embedded();
              then:
                - memory_flasher.write_embedded_image:
      ..... etc ....etc .....etc

that1guy, 2025-05-17:
Hi @rikksullenberger, the code formatting in your original message is making your ticket a bit confusing to read.  Can you reformat your original message so I can understand a little better?

gnumpi, 2025-05-17:
Hey @rikksullenberger , I think I know what causes your issue and I am on it to fix it. In the mean time, can you try to replace the full yaml file in you dashboard editor with the content of this file https://github.com/FutureProofHomes/Satellite1-ESPHome/blob/develop/config/satellite1.dashboard.yaml ?

gnumpi, 2025-05-17:
Another current solution would be to switch over to the beta release before taking control:
1. Delete the current config in your ESPHome builder - dashboard.
2. On you control page of HA enable the beta switch:

[image]
3.  On the top of the 'configuration' section it should now report that a new update is available (might take a bit)
4. Click on that notification and perform the update to the latest beta.
5. Now you should be able to go back to your ESPHome builder and take control of your device.