FutureProofHomes/Satellite1-ESPHome #321: Satellite1 randomly becomes unresponsive.

bumbblebee, 2025-05-21:
## Describe the bug
There are times where Satellite 1, regardless of firmware, becomes completely unresponsive.

### Steps to reproduce
Unable to reproduce error as this is a random issue. It could be a day, three days, or anything that happens in between.

### Expected behavior
When unit becomes unresponsive wake work does not work, mute button only shows a single red light instead of the outer ring lighting up, volume does not show on the rings, and the action button will just sit and spin the entire time. The unit becomes responsive after cycling power.

### Environment
 Currently running the latest  beta version 

### Additional context
Video of the  issue
https://photos.app.goo.gl/htA6BtEHNw7ibEpu9

ShaneLucPicard, 2025-06-01:
I also see this. I have found that many times unplugging and plugging back in doesn't correct the issue maybe about 20% of the time does this work. However if you press the reset button on the CORE board it will start working maybe 50% of the time it's still hit or miss but a little better odds. I am currently on Firmware: v0.1.2+ld2450 (ESPHome 2025.4.0) but this has been the case on all firmware's up to this point.

that1guy, 2025-06-11:
Hello friends.  Please update to our [v1.0.2 release](https://github.com/FutureProofHomes/Satellite1-ESPHome/releases/tag/v0.1.2) as this seems to resolved a lot of stability challenges.  Report back here on your findings!  Thank you!

gnumpi, 2025-06-12:
Thanks for sharing, @lordratner. I wish there was a way to trigger that issue. This would make finding the bug sooo much easier.

bumbblebee, 2025-06-14:
Alright, not even 20 hours later, and I encountered the issue again. And this time I think it's even worse. So, before I could get a response from the buttons, now it wasn't even responsive. The usual trick on cycling the power to it fixed the issue.

bumbblebee, 2025-06-26:
To test, I detached my LD2450, and I still have the same issues; however, I have noticed that the problem is happening less often after the patch.