FutureProofHomes/Satellite1-ESPHome #328: Update failing with missing file

hdprice, 2025-06-20:
When updating the Satellite1 from the ESPHome Builder, it fails with the error

Error reading file /config/esphome/satellite1.base.yaml: [Errno 2] No such file or directory: '/config/esphome/satellite1.base.yaml'

This file is referenced in the YAML config file as an include. 

Everything was working fine until the HA Voice update this week. 

Any pointers, suggestion, help, etc would be very welcome.

hdprice, 2025-06-20:
Well I downgraded to 2025.4.2 and I'm getting the same message. Ugh.

that1guy, 2025-06-21:
Odd.  @gnumpi can you please take a look?

hdprice, 2025-07-18:
> Closing this issue because it's stale, but dropping this issue here. :)
> The Sat1 firmware is not yet compatible with the latest ESPHome versions.  We recommend compiling with ESPHome version 2025.5.2.  Read about how to do this here.
> If you're building from command line, simply use the correct ESPhome version by issuing the command: pip install esphome==2025.5.2
> Feel free and close this issue if this resolved your issue!  Thank you!
> image.png (view on web)
>
>
> All content is personal opinion. All content is copyright. All rights are reserved. No reproduction in any form is allowed without prior written permission

I tried to install the new beta software but I’m unable to because of this problem. I tried downgrading ESPHome but doing broke everything else. LOL. Is there some way to install the beta firmware directly from my Mac, maybe using a USB cable? ThanksHoward