FutureProofHomes/Satellite1-ESPHome #339: Audio cut outs and artifacts

tomfrenzel, 2025-08-04:
## Describe the bug
When playing music over the media player entity in Home Assistant that music starts to cut out after a couple minutes. It starts by making the music "slower" and continues by making quick pauses before continuing to play. After about a minute to music stops completely. It seems like something struggles to process the audio.

### Steps to reproduce
Let music play for approx. 10min

### Expected behavior
The music should keep playing without any artifacts or cutouts

### Environment
- Latest Beta of this firmware
- Home Assistant 2025.7.4
- WiFi Signal is -59dbm on 2.4GHz Channel 1
- Temperature is reported to be 35°C

EuleMitKeule, 2025-09-03:
I have frequent (every ~10-15s) stuttering when playing back any audio either via Home Assistant or Music Assistant on my Satellite1 with the latest beta firmware. Definetely not a hardware issue, the same happens for two different Satellite1s for me.

EuleMitKeule, 2025-10-25:
For me switching from my Netgear Nighthawk MR60 mesh router to any other router fixed the issue.