FutureProofHomes/Satellite1-ESPHome #345: Sat1 unit not powering on or initialising on first start up - Fresh purchase, new out of the box.

mosaau, 2025-08-06:
## Describe the bug
I purchased the Sat 1 and hat and it arrived yesterday. using the provided cable and power adapter I powered on the unit and the 2 5V and 3V red led's were lit. that is all that happened, no startup animation or additional steps like in the setup instructions.

### Steps to reproduce
Brand new out of the box, plug into ESP32/core usbc port with provided cable and power plug.

### Expected behavior
Device is supposed to power up and start an initialisation light animation

### Environment
 - OS: [e.g. Arch Linux]
 - Other details that you think may affect.

### Additional context
Additionally I thought the unit had no pre flashed software so I went to flash using the futureproofhomes ESP software flasher in google chrome, the provided cable and other usbc cables I had around the house, but the JTAG serial connection keeps dropping in and out making it unable to select the port for connection nd flashing. However a generic ESP32 I have laying around the house, using the sat1 provided usbc cable, works perfectly fine.

gnumpi, 2025-08-07:
I just saw that you already have tried flashing the device via usb. Could you try detaching the HAT from the core board, plug the usb-c cable directly to the core board and try running the flashing process again?

asucrews, 2025-08-08:
Just to add to this for tracking propose, i also received what looks like non flashed device. I was able to flash it and get working.

mosaau, 2025-08-08:
> Just to add to this for tracking propose, i also received what looks like non flashed device. I was able to flash it and get working.

Thank you for this reply. What did you do to flash your unit as I am unable to flash mine.
Also what OS were you running on to flash the unit?

asucrews, 2025-08-08:
I just followed this https://docs.futureproofhomes.net/satellite1-flash-via-usb-c/ and after install was done i power cycled it?

BaseballSS, 2025-08-08:
Mine also appeared to arrive not flashed. I was able to follow https://docs.futureproofhomes.net/satellite1-troubleshooting/?h=boot#force-the-core-board-into-boot-mode to force it into boot mode and https://docs.futureproofhomes.net/satellite1-flash-via-usb-c/ to flash it. FWIW, for both forcing into boot mode and flashing I followed the steps of powering the CORE board directly.

KrX3D, 2025-08-09:
> I just followed this https://docs.futureproofhomes.net/satellite1-flash-via-usb-c/ and after install was done i power cycled it.

Thats exactly what i did, i also received a unit like 10 days ago. I tried to follow the "tutorial" on youtube -> https://youtu.be/WeOEl3jho1k?t=652 
but got stuck at the part where the leds sould turn blue and it should start flasing the xmos. for me that didnt happen but i just saw the white blinking like shown here the step after: https://youtu.be/WeOEl3jho1k?t=665

that1guy, 2025-08-12:
> I experienced the same issue today after receiving my unit. I first tried flashing via the HAT while it was connected to the core ESP, but ran into intermittent USB dropouts. Resetting with the right push button seemed to fix the dropout issue, allowing me to flash successfully — but the process stopped short of completing Wi-Fi setup.
> 
> I then removed the core and flashed it separately, which completed successfully and connected to Wi-Fi without issues. However, after reinstalling the HAT, it had no functionality, and I also lost functionality of the core.

Hey Chris, I just emailed you a tracking number with a new HAT and CORE.  Let me know how those units behave.

gnumpi, 2026-01-14:
@daviddever, your issue sounds like a connection problem between core and hat. Can you confirm that the core is pushed into the hat properly? You should be able to see the top of the pins from the other side (https://docs.futureproofhomes.net/satellite1-assembly/). 
You should always power your device via the CORE/ESP port, only that one has the power delivery capability. 
Your dmsg and lusb logs from the XMOS port indicate that the firmware is already successfully flashed to the XMOS. Without it  there wouldn't be any usb service available at all.

jimrkerry, 2026-10-15:
Got my Satellite1.1 HAT board in the spring of 2026.  Using it with a Raspberry Pi Zero 2 W instead of the Core Board and that has been working well running the Linux Voice Assistant (LVA) software.

Noticed recently that someone had created a Docker image to run along side the LVA that would control the LEDs and buttons on the Satellite1.1 HAT board.  Located here: https://github.com/OHF-Voice/linux-voice-assistant/tree/main/examples/Satellite1%20HAT%20Board

The Docker image to control the LEDs and buttons runs without errors but the LEDs stay dark and the buttons don't work.  I thought maybe the HAT board needed to be flashed.  I tried connecting the HAT board with and without the Pi Zero 2 attached to two different computers using several different USB cables and the HAT board would never connect.

I suspect I am having the same issue as others here and didn't realize it at the time.  The Satellite1.1 HAT board works well with LVA as a Home Assistant satellite.  Just that I have never seen the LED ring light up or a button work.  Would like to know if there is any way to get those things working.