FutureProofHomes/Satellite1-ESPHome #348: [i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE

Godzookee, 2025-08-15:
## Describe the bug
No response to voice commands
### Steps to reproduce
Happens during setup

### Expected behavior
Should made audible tone and respond to voice comands

### Environment
 - OS: Windows 11
Core 2025.8.1
Supervisor 2025.08.1
Operating System 16.1
Frontend 20250811.0

### Additional context
Have tried reflashing multiple times both in and out of the HAT.  Device is detected in ESPHome and options at there however now audio is received or acknowledgement of wake word.  Using supplied USB-C cables and power adapter.

gnumpi, 2025-08-15:
Hey, sorry to hear that you are having troubles with setting up your satellite1. This error you are seeing usually happens when the ESP doesn't get its I2S clock signal (It is provided by the XMOS chip on the HAT). Can you double check that your HAT and CORE boards are properly attached to each other? Can you also check if the LED ring is indicating the change when pressing the volume up and down buttons? Would you mind sharing the full log file?

Godzookee, 2025-08-15:
`
[17:10:48][D][wifi:545]: - 'SG-B21****7800'[redacted] (AC:1F:09:01:AA:5C) [redacted]▂▄▆█
[17:10:48][I][wifi:313]: WiFi Connecting to 'XXXXXXX_Iot'...
[17:10:48][D][esp-idf:000][wifi]: W (10251) wifi:
[17:10:48][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[17:10:48][D][esp-idf:000][wifi]: 
[17:10:48]
[17:10:49][D][esp-idf:000][wifi]: I (10619) wifi:
[17:10:49][D][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[17:10:49][D][esp-idf:000][wifi]: 
[17:10:49]
[17:10:49][D][esp-idf:000][wifi]: I (10621) wifi:
[17:10:49][D][esp-idf:000][wifi]: state: init -> auth (b0)
[17:10:49][D][esp-idf:000][wifi]: 
[17:10:49]
[17:10:49][D][esp-idf:000][wifi]: I (10624) wifi:
[17:10:49][D][esp-idf:000][wifi]: state: auth -> assoc (0)
[17:10:49][D][esp-idf:000][wifi]: 
[17:10:49]
[17:10:49][D][tas2780:474]: Current Mode: SOFTWARE_SHUTDOWN (PWR_MODE: 2)
[17:10:50][D][esp-idf:000][wifi]: I (11626) wifi:
[17:10:50][D][esp-idf:000][wifi]: state: assoc -> init (2700)
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11627) wifi:
[17:10:50][D][esp-idf:000][wifi]: Coexist: Wi-Fi connect fail, apply reconnect coex policy
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11629) wifi:
[17:10:50][D][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][W][wifi_esp32:711]: Event: Disconnected ssid='XXXXXXX_Iot' bssid=F6:92:BF:95:50:F9[redacted] reason='Unspecified'
[17:10:50][W][wifi:653]: Error while connecting to network.
[17:10:50][D][wifi:697]: Retrying with hidden networks...
[17:10:50][I][wifi:313]: WiFi Connecting to 'XXXXXXX_Iot'...
[17:10:50][D][esp-idf:000][wifi]: W (11643) wifi:
[17:10:50][D][esp-idf:000][wifi]: Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11649) wifi:
[17:10:50][D][esp-idf:000][wifi]: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11650) wifi:
[17:10:50][D][esp-idf:000][wifi]: state: init -> auth (b0)
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11654) wifi:
[17:10:50][D][esp-idf:000][wifi]: state: auth -> assoc (0)
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11660) wifi:
[17:10:50][D][esp-idf:000][wifi]: state: assoc -> run (10)
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11680) wifi:
[17:10:50][D][esp-idf:000][wifi]: connected with XXXXXXX_Iot, aid = 34, channel 1, BW20, bssid = f6:92:bf:95:50:f9
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11681) wifi:
[17:10:50][D][esp-idf:000][wifi]: security: WPA2-PSK, phy: bgn, rssi: -48
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: I (11682) wifi:
[17:10:50][D][esp-idf:000][wifi]: pm start, type: 1
[17:10:50]
[17:10:50][D][esp-idf:000][wifi]: 
[17:10:50]
[... 412 lines cut ...]
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][D][tas2780:474]: Current Mode: SOFTWARE_SHUTDOWN (PWR_MODE: 0)
[17:11:21][E][tas2780:391]: TDM Clock Error!
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:21][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][D][tas2780:474]: Current Mode: SOFTWARE_SHUTDOWN (PWR_MODE: 0)
[17:11:22][E][tas2780:391]: TDM Clock Error!
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[17:11:22][W][i2s_audio.speaker:142]: Error writing to I2S: ESP_ERR_INVALID_SIZE

Godzookee, 2025-08-15:
Hi Gnumpi

Thanks for the response.  Pressing the up and down on the speaker buttons changes nothing on the LED Ring

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/21792404/esp-web-tools-logs.txt)

Appreciate the assistance.

Godzookee, 2025-08-15:
yes thats right

gnumpi, 2025-08-15:
yeap, the question remains why it is not flashing at boot up. You are logging via USB using https://web.esphome.io ?
It would be interesting to get the logs directly from the beginning after booting. Can you start the logging and then press the "Restart Sat1" in HA diagnostics? And share your logs?

Godzookee, 2025-08-15:
Logs as requested

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/21794585/esp-web-tools-logs.txt)

gnumpi, 2025-08-22:
I assume you are in contact with Brad for a replacement. Will close this issue.

closed 2025-08-22 by gnumpi