FutureProofHomes/Satellite1-ESPHome #349: Satellite1 sound output on Sonos speakers

CosminFRC, 2025-08-16:
Hello.
Can an option be added to select sound output on another speaker/media player? Like, I already have a bunch of Sonos speakers spread throughout the house, in every room, and I would like my voice assistant to respond on the existing speaker in the room where the Satellite1 is and not need another speaker connected to the Satellite1.
It would be great for those who already have speakers.
Thanks, 
Cosmin

that1guy, 2025-08-17:
I agree. This is a highly requested feature we need to look into.

CosminFRC, 2025-08-17:
Great, I can't wait for this feature. I know it's possible to output sound from the Atom Echo to another media player, I was wondering if something similar works on the Satellite1.

chpego, 2025-08-17:
> Great, I can't wait for this feature. I know it's possible to output sound from the Atom Echo to another media player, I was wondering if something similar works on the Satellite1.

Yes, it's possible. I just did it, but you have to “take control” of Sat1 via ESPHome and add these lines of code:
``
voice_assistant:
  on_tts_end:
     - homeassistant.action:
         action: media_player.play_media
         data:
           entity_id: media_player.YOUR # <- change this
           media_content_id: !lambda 'return x;'
           media_content_type: music
``

CosminFRC, 2025-08-22:
> very good question, I must admit I forgot to test it.

Did you test? Mine doesn't resume.

gnumpi, 2025-08-22:
Any recommendations on how we want to select the speaker to which it should be forwarded to? I mean it is already possible if you take control as @chpego has posted. I don't see how we can define that in a pre-compiled firmware though.
Isn't there a way on achieving this purely on HA side? If not, I would rather see this to be a feature request for HA to support setting the input and output of the assist pipeline independently instead of assuming it's always the same device.

gnumpi, 2025-08-22:
> > > Great, I can't wait for this feature. I know it's possible to output sound from the Atom Echo to another media player, I was wondering if something similar works on the Satellite1.
> > 
> > 
> > Yes, it's possible. I just did it, but you have to “take control” of Sat1 via ESPHome and add these lines of code:
> > voice_assistant:
> > on_tts_end:
> > - homeassistant.action:
> > action: media_player.play_media
> > data:
> > entity_id: media_player.YOUR # <- change this
> > media_content_id: !lambda 'return x;'
> > media_content_type: music
> 
> But if you are playing music on that media player, will it pause and resume what was playing before the TTS?

you can try adding `announce: true` to the data section, but it depends on the speaker you are forwarding it to if it supports it.

that1guy, 2025-08-29:
@gnumpi Can we not use this trick as a viable solution:  

https://esphome.io/components/api/#api-device-actions
https://esphome.io/components/api/#user-defined-actions
https://discord.com/channels/1242282357298298941/1242282358212792402/1410400077104681103

baudneo, 2025-08-31:
Just wanted to voice my support for this. It would be nice if HASS had a way to ask for all current media players and return a dict of them and associated data, so a select component could be populated with parsed results (maybe esphome could get that implemented and expose it, as it would be beneficial for their voice assistant as well).

The user defined actions could work in the interim by asking the user to set a media_player entity to broadcast to, stored as a global with flash retention. It would be nice to have some sort of component to use rather than hooking into `tts_end` that would allow streaming responses to a media player instead of/alongside the audio/spk out that will handle ducking on that external media player. I have heard ChimeTTS allows for ducking across all sorts of different media players, but I haven't tested it yet.

that1guy, 2025-11-21:
Hi everyone, this feature request should be solved at the Home Assistant level and not at the Satellite1 firmware-level.  Would anybody here be interested in developing a HA Blueprint that will allow to:

1. Select a Sat1 or any voice assistant in HA
2. Select the media play that HA should route the audio responses to

We could then add this blueprint to our repo here: https://github.com/FutureProofHomes/Satellite1-HA-Automations

This implementation may require some other hackery, but if necessary we could ask Nabu for this feature?

CosminFRC, 2025-12-01:
> I can take a look this weekend, I haven't even implemented it for myself yet and don't use my sat1 as a result. So I'll light a fire to get my local VA working and lay it out in a blueprint.

That would be great. Thanks.

GL75, 2026-02-08:
***WORKAROUND***
Went down a rabit hole for the last 36 hours. Tried to bring this back at the home assistant layer by having an automation trigger based on events on the wyoming pipeline without any luck, extremely frustrating.

As a workaround, started to have some success by bringing it back to the ESP32 and leveraging MQTT.

Using `on_tts_end` I was able to send the URL of the flac file over MQTT and have an automation triggered by listening to MQTT topic.

The caveat here is that I want to leverage the sonos cloud integration for announcement as it is the most elegant way to take care of these. Yes there's a cloud dependency that I am trying to avoid but this work very elegantly. A battle for another day.

Thought I was there until I learn that the sonos cloud integration does not support .flac files. That was peak level frustration.

So finally leveraging the same approach but one step earlier at `on_tts_start` I was able to pass the actual string prior to the TTS processing. An automation is monitoring on the HA side the MQTT topic and trigger a tts action to a Sonos cloud entity

Code to add on the esp32 : 
```
mqtt:
  broker: 192.168.0.xxx # Your MQTT Broker IP
  username: "mqtt"
  password: "strongpassword"

voice_assistant:
  on_tts_start:
    - mqtt.publish:
        topic: "home/voice/response_text"
        payload: !lambda |-
          return "{ \"reply\": \"" + x + "\", \"device\": \"living_room_sat\" }";
```

HA Automation:
```
alias: Voice Redirect
description: ""
triggers:
  - trigger: mqtt
    options:
      topic: home/voice/response_text
conditions: []
actions:
  - variables:
      data: "{{ trigger.payload | from_json }}"
      room_map:
        kitchen_sat: media_player.kitchen_announce
        living_room_sat: media_player.living_room_ceiling_announce
      target_player: "{{ room_map.get(data.device, 'media_player.living_room_sonos_cloud') }}"
    enabled: true
  - action: tts.cloud_say
    metadata: {}
    data:
      cache: false
      entity_id: "{{ target_player }}"
      message: "{{ data.reply }}"
mode: single
```

Still a bit of clean up to do but this does exactly what I wanted. A workaround that works very well until a proper solution exist