FutureProofHomes/Satellite1-ESPHome #350: voice assistant works great, music assistant doesn't pkay

sdetweil, 2025-08-17:
## Describe the bug
i have a new fh satellite working w latest HA 
get speech output from voice requests

music assistsnt sees device as player, can select device to play music (from spotify)
get play now/plsy button , but it doesnt work

thru phone works great. so the MA side looks right up to the streaming part

i have 4 roku that are seen too, but that pkayer doesnt exist there

### Steps to reproduce
Steps to reproduce the behavior.
ibatall, flash firmware, add thru esphome (auto discovered) . config assistant
test assistsnt

add as player to music assistant.
select player
select music
play now

### Expected behavior
expected it to work

### Environment
 - OS: ha & ma in docker on ubuntu linux
 - NOT HAOS , ma installed manually as addons not present in my deployment

 - Other details that you think may affect.

this works to iphone

dlk57, 2025-08-20:
I have the exact same issue with sat 1's not playing through music assistant. Others such as Denon work fine with MA. Don

sdetweil, 2025-08-20:
ok, the streamserver setting used the docker container address, not the docker host address...  so the player couldn't connect

closed 2025-08-20 by sdetweil