FutureProofHomes/Satellite1-ESPHome #352: Add option to manually configure Snapcast server host/IP and port

ret1, 2025-08-22:
## Summary
Enable explicit configuration of the Snapcast server host/IP and port.

### Motivation
In environments with multiple IP networks/VLANs, automatic discovery does not reach all subnets. My satellite1 resides on a different network/VLAN, which currently prevents using it via Snapcast in Music Assistant. Allowing manual entry of the Snapcast server host/IP and port would restore functionality in multi-subnet setups.

that1guy, 2025-08-22:
@ret1 I believe this is possible today by supplying the IP address of the SnapServer in your yaml.  For example:

```
snapcast:
  id: snapcast_client
  server_ip: 192.168.178.101
```

Follow these instructions to modify the Sat1 firmware: https://docs.futureproofhomes.net/satellite1-modifying-the-firmware/#import-the-sat1-to-your-esphome-dashboard

@gnumpi can we get this commented out yaml to appear in the the ESPHome Dashboard Builder by default so the user can quickly add the optional SnapServer ip address, recompile, flash and be done?

gnumpi, 2025-08-22:
> [@gnumpi](https://github.com/gnumpi) can we get this commented out yaml to appear in the the ESPHome Dashboard Builder by default so the user can quickly add the optional SnapServer ip address, recompile, flash and be done?

yes sure I will add this to the next beta update! I am also planing to let media player switch its snapcast server if it receives an url of the form: `snapcast://192.168.178.101:1704/Music%20Assistant%20-%20snapcast4e19f0?RPC_PORT=1705` but this will not yet be included in the next beta update.

gnumpi, 2025-08-27:
done. The template for manually setting the snapcast server is now included in the dashboard yaml file

closed 2025-08-27 by gnumpi