FutureProofHomes/Satellite1-ESPHome #362: Spamming my HA log

chezpaul, 2025-09-05:
## Describe the bug
My log is just spammed all day none stop with
```
2025-09-04 20:21:01.142 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:02.171 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:03.189 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:04.207 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:05.235 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:06.258 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:07.274 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:08.295 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:09.311 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:10.329 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:11.348 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
2025-09-04 20:21:12.377 WARNING (MainThread) [homeassistant.components.esphome.manager] Satellite1 ce2dd8: [W][ltr_als_ps:472]: Sensors blacked out
```

### Steps to reproduce
Restart doesn't change a thing

### Expected behavior
No spamming

### Environment
 - OS: [e.g. Arch Linux]
 - Other details that you think may affect.

### Additional context
Add any other context about the problem here.

Brad in the chat room said it's because my sensor is in the dark which it is since I work in a dark recording studio. Is there a way to not have this logged?

redneckzking, 2026-01-17:
Music constantly stops playing every 15-20 seconds on both of my Sat 1 devices.

Logger: homeassistant.components.esphome.manager
Source: components/esphome/manager.py:520
integration: ESPHome (documentation, issues)
First occurred: 23:03:39 (306 occurrences)
Last logged: 23:38:42

Satellite1 d09ff0: [W][component:456]: Components should block for at most 30 ms
Satellite1 d09ff0: [W][component:453]: gpio.binary_sensor took a long time for an operation (65 ms)
Satellite1 c414fc: [W][ltr_als_ps:472]: Sensors blacked out
Satellite1 c414fc: [W][ltr_als_ps:387]: Data available but not valid
Satellite1 c414fc: [W][ltr_als_ps:176]: Can't get data after several tries.

glbailey, 2026-01-20:
+1
Logger: homeassistant.components.esphome.manager
Source: components/esphome/manager.py:520
integration: ESPHome ([documentation](https://www.home-assistant.io/integrations/esphome), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+esphome%22))
First occurred: January 18, 2026 at 9:15:37 PM (15660 occurrences)
Last logged: 8:44:06 AM

    Satellite1 d0fa0c: [W][ltr_als_ps:169]: Can't get data after several tries.
    Satellite1 d0fa0c: [W][component:157]: Component ltr_als_ps.sensor set Warning flag: unspecified
    Satellite1 d0fa0c: [W][ltr_als_ps:465]: Sensors blacked out
    Satellite1 d0fa0c: [W][component:172]: Component ltr_als_ps.sensor cleared Warning flag
    Satellite1 d0fa0c: [W][ltr_als_ps:485]: Impossible ch1/(ch0 + ch1) ratio